Zanduino/INA #6: documentation: more clearly indicate how to clear interupt flag

dnlwgnd, 2018-07-10:
Hi there,

working with the lib I literally just spend over an hour looking for a bug until I realised that when using OnConversion-Alert, I have to call waitForConversion to clear the Interrupt in the device. This wasn't intuative for me as I thought I replaced the waitForConversion usage with an interrupt handler.
I was somehow expecting, that reading the data registers does reset the flag.
I assume calling waitForConversion in this case does not acutually wait and loose any time.

Maybe this case could be made clearer in the documentation of alerts and the waitForConversion-Function.

cheers, Daniel.

SV-Zanshin, 2018-07-11:
I'm looking at the wiki documentation and the https://github.com/SV-Zanshin/INA/wiki/alertOnConversion() page describes the mechanism pretty clearly, 

> ...The state will remain set until either the configuration register is read or the Mask/Enable register is read. The simplest means to reset the state is to call waitForConversion() which will ...

The https://github.com/SV-Zanshin/INA/wiki/waitForConversion() page has a final line 

> ...It also resets the flag so that it can be set by the next conversion.

But I will change that to break it out into a separate paragraph so that it pops out a bit more.

If you think that this isn't sufficient, could you suggest some new wording which I could add?

SV-Zanshin, 2018-07-11:
Added paragraph to https://github.com/SV-Zanshin/INA/wiki/waitForConversion()

closed 2018-07-11 by SV-Zanshin

dnlwgnd, 2018-07-11:
cool. now i obvioiusly did not read everything carefully enough :-)
but with the new paragraph I am more than happy.
oh btw, thank you for the great library.

SV-Zanshin, 2018-07-11:
Glad you like the library and thanks for contributing! I'm using this library for a mixed environment of INA devices and I'm glad someone one can benefit from it.