Zanduino/INA #27: Issues with ESP32

delboy711, 2018-09-04:
I found two issues with using the INA library with an ESP32 processor.

First
-----
ESP32 allows programmable selection of I2C pins and while pins 21,22 are the default, those pins do not appear on many development boards. My workaround was to hardwire SDA, and SCL into the Wire.begin statement in INA.begin.

Second
---------
INA.begin tests to see how much space is available in the EEPROM.  But ESP32 returns EEPROM.length as zero by default, and begin fails.  My workaround was simply to skip the test.

SV-Zanshin, 2018-09-15:
I've just ordered a 32U4 Arduino so that I can test the library with that hardware; then I'll see about getting a permanent solution

SV-Zanshin, 2018-09-15:
Oops - I just realized that I didn't want any more 32U4 boards (I already have a couple), but needed the ESP32 from Espressif instead. I don't know if I'm ready to do that quite yet, but I have downloaded the board from Espressif and will see if I can reproduce/fix this issue with just the compiler.

SV-Zanshin, 2018-09-20:
The ESP32 just arrived and I'll do the testing this weekend.

SV-Zanshin, 2018-09-22:
I've changed the code to work with ESP32 based devices, specifying a hardcoded 512 Bytes of virtual EEPROM memory.

Regarding the I2C part of your bug report, could you see if adding

SV-Zanshin, 2018-09-22:
Changes to the I2C SDA/SCL setup is too specific for the general library, and adding it to the begin() method as optional values causes issues with the current functionality, so I will close this as a bug although we might open it as an enhancement.

closed 2018-09-22 by SV-Zanshin