Zanduino/INA #28: INA226 Current Reading Saturation

007trains, 2018-09-19:
When setting up a 75A 75mv shunt using INA.begin(75,1000) the current reading saturates at 10.999A
Trialing INA.begin(31,1000)  works correctly at 30.999A
Trialing INA.begin(32,1000) saturates at 0A
Trialing INA.begin(33,1000) saturates at 0.999A
Its possible there is a variable that is reaching its limits and rolling over in the INA.begin() function causing the wrong maxCurrent to be written
I can work around this issue by changing INA.begin(75,1000) to INA.begin(7.5,10000) then multiplying the current by 10

SV-Zanshin, 2018-09-19:
It does look like an overflow issue. I will look into it.

SV-Zanshin, 2018-09-19:
I have found the cause of the issue. In order to save memory the values in the "inaEEPROM" structure are not stored as full bytes but as a given number of bits. The "maxBusAmps" values is stored in 5 bits, giving a range of 0-31A before overflowing. I designed the library with my intended purposes in mind and always had just a couple of amps to measure, so didn't think about documenting the 31A limit.

Adding another 2 bits to the structure would allow currents of up 127A to be measured.

SV-Zanshin, 2018-09-19:
I've fixed the issue and committed the new fix

closed 2018-09-19 by SV-Zanshin