Zanduino/INA #29: I2C connection  in not working on Wemos D1 (Esp8266) + INA226

ukrduino, 2018-10-12:
HI Arnd!
I have [INA226](https://ru.aliexpress.com/item/Free-shipping-CJMCU-226-INA226-IIC-interface-Bi-directional-current-power-monitoring-sensor-module/32741230177.html) and [Wemos D1 mini](https://ru.aliexpress.com/item/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-by/32651747570.html). They are propperly connected and working [with example code](https://github.com/1technophile/OpenMQTTCharger/blob/master/OpenMQTTCharger.ino)

but when i try to use your example DisplayReadings i can make them connect.
I have looked though code of INA.cpp in place:
```c
if (_DeviceCount==0)                                                        // Enumerate devices in first call  //
  {                                                                           //                                  //
    #ifndef ESP8266                                                           // I2C begin() on Esplora problems  //
      Wire.begin();                                                           // Start I2C communications         //
    #endif                                                                    //                                  //
    #ifdef __STM32F1__                                                        // Emulated EEPROM for STM32F1      //
      uint8_t maxDevices = EEPROM.maxcount() / sizeof(inaEE);                 // Compute number devices possible  //
    #else                                                                     // EEPROM Library V2.0 for Arduino  //
      uint8_t maxDevices = EEPROM.length() / sizeof(inaEE);                   // Compute number devices possible  //
    #endif                                                                    //                                  //
    #ifdef ESP32                                                              //                                  //
      EEPROM.begin(512);                                                      // If ESP32 then allocate 512 Bytes //
      maxDevices = 512 / sizeof(inaEE);                                       // and compute number of devices    //
    #endif                                                                    //                                  //
    for(uint8_t deviceAddress = 0x40;deviceAddress<0x80;deviceAddress++)      // Loop for each possible address   //
    {                                                                         //                                  //
      Wire.beginTransmission(deviceAddress);       ```

seems it does everything correct but still no connection ... 
Could you advice something, please.

ukrduino, 2018-10-12:
https://github.com/jarzebski/Arduino-INA226/tree/master/INA226_simple - this example works as well.

SV-Zanshin, 2018-10-13:
Greetings!

 

The ESP8266 doesn’t have fixed pins for SDA and SCL, so one of the changes made was to comment out the initialization of the Wire library (the “Wire.begin()” call) to allow the user to define their own pins inside their setup() method.

 

Can you try to call “Wire.begin([your_SDA_Pin],[your_SCL_Pin]);” in your main program setup() method? I don’t have an ESP8266  handy, but I believe the default is “Wire.begin(2,14);”

 

Thanks,

 

-Arnd.

SV-Zanshin, 2018-10-13:
If that works I'll update the code for both ESP8266 and the ESP32 and update the documentation to make sure that the developer issues a call to "Wire.begin()" in their setup() function.

ukrduino, 2018-10-13:
I will try and reply soon, thank you!

SV-Zanshin, 2018-10-20:
I've modified the wiki page https://github.com/SV-Zanshin/INA/wiki/begin() and changed the code so that the developer needs to issue the appropriate Wire.begin() call prior to calling the library.

closed 2018-10-20 by SV-Zanshin