Zanduino/INA #30: Using INA in Teensy 3.x boards

AluminioVanadio, 2018-10-14:
I have been unable tu use the library in a teensy 3.2 board, I changed the include from Wire.h to i2c_t3.h. It compiles but at least the example file does not detect an INA226. I can detect it using a scanner sketch which runs OK in teensy.

SV-Zanshin, 2018-10-14:
Does one of the example programs work when using the standard “Wire.h” library?

 

-Arnd.

SV-Zanshin, 2018-10-15:
> Hi,
> I tried BackgroundRead also, with the library freshly installed and does not compile :
>
> Compiling 'BackgroundRead' for 'Teensy 3.2 / 3.1'
>
> BackgroundRead.ino: 71:5: error: expected constructor, destructor, or type conversion before '(' token
> ISR (PCINT0_vect) { \ handle pin change interrupt D8
>
> BackgroundRead.ino: In function void setup()
> Error compiling project sources
> Build failed for project 'BackgroundRead'
>
> BackgroundRead.ino: 95:3: error: 'PCIFR' was not declared in this scope
> PCIFR |= bit (digitalPinToPCICRbit(INA_ALERT_PIN)); \ clear any outstanding interrupt
>
> BackgroundRead.ino: 96:3: error: 'PCICR' was not declared in this scope
> PCICR |= bit (digitalPinToPCICRbit(INA_ALERT_PIN)); \ enable interrupt for the group
>
> Using Visual Studio IDE.
>
> Compiles OK selecting an Arduino Leonardo. It looks like port manupulation in teensy is different. Also I don not hace interrupts pins hooked up.
> DisplayReadings compiles but it does not detect the INA226.
> No INA device found, retrying in 10s...
> No INA device found, retrying in 10s...
> No INA device found, retrying in 10s...
>
> SDA and SCL pins are correctly mapped by Wire.h looking at the scope.

Hello,

 

Any code that revolves around interrupts and interrupt handling is going to be difficult to get running on all processors. The programs were developed and tested on Atmega 8 bit boards so I’m not surprised that things like the “PCIFR” and “PCICR” buffers don’t match up.

 

But the base library should work, and the “DisplayReadings” program is pretty basic and doesn’t use interrupts so it should work. The EEPROM handling looks to be same as well, which was a problem for Esplore devices.

 

Could you try to add the following green line in the library code at Line 193 of the “INA.cpp”?

 

     Serial.print(“Max Devices is “);Serial.println(maxDevices);

     Serial.print(“Found device at “);Serial.println(deviceAddress);

     if (Wire.endTransmission() == 0 && _DeviceCount < maxDevices)           // If no error and EEPROM has space //

      {                                                                       //                                  //

        originalRegister = readWord(INA_CONFIGURATION_REGISTER,deviceAddress);// Save original register settings  //

        writeWord(INA_CONFIGURATION_REGISTER,INA_RESET_DEVICE,deviceAddress); // Forces INAs to reset             //

        tempRegister     = readWord(INA_CONFIGURATION_REGISTER,deviceAddress);// Read the newly reset register    //

 

 

My guess is that the “EEPROM.length()” function might not be working correctly for that device and we would need to add a device-specific “#ifdef” to the code.

 

-Arnd.

AluminioVanadio, 2018-10-15:
Hello Arnd.
I added the code, and maxDevices is at 0. I forced maxDevices to 127 and it works OK, you are right, there are differences in the EEPROM implementation of the Teensy.
Thanks

SV-Zanshin, 2018-10-16:
I added some #ifdefs to support the TEENSY boards to the code, that should work for your implementation now.

AluminioVanadio, 2018-10-17:
Line 188 of the modified has the variable type declaration missing:
 **uint8_t** maxDevices = 2048 / sizeof(inaEE);      
But stiil doesn't work in teensy 3.2, unless I assign maxDevices a fixed value.

SV-Zanshin, 2018-10-17:
That is quite odd. Did you get a compiler error with the missing type declaration or did you notice it yourself? I am asking because perhaps the “CORE_TEENSY” isn’t the correct variable to check.

SV-Zanshin, 2018-10-18:
The compiler error would seem to indicate that the branch in the #ifdef was taken correctly, but I've  added the directives you listed above to the code and fixed the way that the variable "maxDevices" is declared. Can you check to see if the current version works correctly?

AluminioVanadio, 2018-10-18:
Hi,
The problem is that 2048/ sizeof(inaEE) is 256, and maxDevices is 8 bit long, so making maxDevices a 16 bit integer or limiting maxDevices to 255 should work.
I modified it to:
maxDevices = 2048 / sizeof(inaEE)-1; 	  		
and works fine.

SV-Zanshin, 2018-10-19:
Thanks for pointing out that error - I should start using those bigger boards to broaden my horizons from the much more resource-restricted Atmel processors.

closed 2018-10-19 by SV-Zanshin