Zanduino/INA #35: getBusMicroAmps giving 2x actual current on a INA226

bramhut, 2019-01-10:
It appears that the function `getBusMicroAmps` is giving a current value which is 2x higher than the actual current measured with a multimeter. I'm using a cheap INA226 breakout board from China which has a built-in resistor of 10 mOhm. 
I have also tested it with the DisplayReadings example with the 10A, 10 mOhm settings and the problem still occurs.

Did I overlook something or is it a bug?

Thanks for the great Library btw!

SV-Zanshin, 2019-01-11:
Let me see if I can reproduce the problem.

SV-Zanshin, 2019-01-12:
Hello bramhut,

could you change your program to output both the "getBusMicroAmps()" value as well as the corresponding "getShuntRaw()" value?  If the raw shunt value doesn't correspond to the expected value then perhaps the built-in resistor isn't 10mOhm which would explain an incorrect reading. If the raw shunt value is correct, then something is wrong in the code. 

p.s. Could you also post your "INA.begin()" call?

bramhut, 2019-01-14:
Hello,

I have checked the the getShuntRaw() values and they are still 2x as high as they should be with a LSB of 2.5uV. 
Here is debug code I'm running:
```
#include <INA.h>

INA_Class INA;

void setup()
{
	INA.begin(10,10000,0x40);
	INA.setMode(INA_MODE_CONTINUOUS_BOTH);
	Serial.begin(115200);
	Serial.println(INA.getDeviceName());
	delay(1000);
	INA.setShuntConversion(41560, 0x40);
	INA.setBusConversion(41560, 0x40);
	INA.setAveraging(16, 0x40);
	pinMode(9,OUTPUT);
}

void loop()
{

	int32_t busMillivolts = INA.getBusMilliVolts();
	int32_t busMicroAmps = INA.getBusMicroAmps();
	int32_t busMicroWatts = INA.getBusMicroWatts();
	int32_t busShuntRaw = INA.getShuntRaw();
	int32_t shuntMicroVolts = INA.getShuntMicroVolts();
	Serial.print("U: ");
	Serial.print((float) busMillivolts / 1000, 2);
	Serial.println(" V");
	Serial.print("I: ");
	Serial.print((float) busMicroAmps / 1000, 2);
	Serial.println(" mA");
	Serial.print("P: ");
	Serial.print(busMicroWatts / 1000);
	Serial.println(" mW");
	Serial.print("Raw: ");
	Serial.println(busShuntRaw);
	Serial.print("Microvolts: ");
	Serial.println(shuntMicroVolts);
	delay(2000);

}
```
After some measuring with my multimeter, I concluded that the resistor is actually a 10 mOhm resistor. And the shunt voltage values that the INA returns are exactly 2x as much as my multimeter reads.

SV-Zanshin, 2019-01-15:
Thanks for the information. Could you also add the output you are getting from this program and what the actual volts/amps are at the time as measured by your multimeter?

Thanks

bramhut, 2019-01-15:
I think I found out what the problem was. The traces from IN+ and IN- are only 1.6mm wide and with a total trace length of around 4cm from IN+ to IN- this will add a resistance of 12mOhm. I conducted some tests and I found out that the voltage across IN+ and IN- is around 55.47mV with a current of 2.44A. This comes down to 22.73 mOhm of total resistance which is actually very close to the 10mOhm resistor + 12mOhm calculated resistance of the traces.
To reduce the problem I could have used the bigger solder pads close to the resistor instead of the IN+ and IN- pins but this would still add some resistance. The better solution would be to use an external shunt and desolder the existing one.

Thanks for the help!

SV-Zanshin, 2019-01-15:
Thanks for the update and debugging information. I tried to reproduce the problem here with my "test bed" and couldn't, which is why I asked for the additional information. I'm glad that you had the expertise and experience to get to the root of the problem yourself, otherwise it would have been difficult indeed to solve this problem!