Zanduino/INA #38: getId() to return dec/Hex value of the Id

novski, 2019-02-08:
**Is your feature request related to a problem? Please describe.**
i can not allways determine which INA219 is read as Id's are incrementaly created.

**Describe the solution you'd like**
I wold like to see the example displayreadings.ino to show the Hexadecimal or Decimal id of the read shunt.

**Describe alternatives you've considered**
Don't know if thats possible. No alternatives. 

**Additional context**
[image]

[image]

SV-Zanshin, 2019-02-08:
That should be possible to add. Let me look into that.

SV-Zanshin, 2019-02-09:
I have added a new library method, "getDeviceAddress()" which returns the device's I2C address. I've also updated the DisplayReadings.ino example program to include displaying the address. Unfortunately, I cannot find my breadboard which has all the different I2C devices installed for testing.

Although this change was trivial, I would appreciate it if you could do the test to see if it functions correctly and post the results to this issue.

Thanks.

SV-Zanshin, 2019-02-09:
I've also changed the comments around in the library, moving all the doxygen comments from the .h to the .cpp

SV-Zanshin, 2019-02-10:
I found my INA226 lying around and checked the changes.

closed 2019-02-10 by SV-Zanshin

novski, 2019-02-10:
Hi, im sorry i was not available shortly. 
I now checked on my side as well and it works great. 
Decimal Address is the best choice! Thanks a lot!