Zanduino/INA #41: [ESP8266] Conflict if EEPROM is used in main code.

miky2k, 2019-05-21:
I use in my esp8266  emulated EEPROM in main code and begin it with (4096) in size for main 
code purpose, i possible to  avoid in  library eeprom size call in library  and add start pointer so i can initialize bigger area and put ina2xx data upper o lower main code data?

miky2k, 2019-05-21:
EEPROM.put((deviceNumber *sizeof(inaEE))+1024,inaEE);  

EEPROM.get((deviceNumber *sizeof(inaEE))+1024,inaEE);

Should put data at 1024  start address.

SV-Zanshin, 2019-05-21:
Adding the 1024 byte offset would break all other implementations, so making that change to the base software is not possible.

It might be possible to change the class instantiation to have an optional offset for the beginning of the EEPROM used by the library, e.g. `INA_Class INA(1024); // instantiate the class with EEPROM offset 1Kb`

miky2k, 2019-05-21:
And if offset is specified in istantiation no EEPROM.begin should be done in library.

SV-Zanshin, 2019-05-21:
At the moment, the INA library initializes with a EEPROM.begin(512) for the 8266. So in order to remain backwards compatible, I wouldn't want to change this. I would expect that if the library has already been initialized a second call won't re-initialize it. Can you check the 8266 code to see what would happen with a second call to begin()?

miky2k, 2019-05-22:
``void EEPROMClass::begin(size_t size) {
  if (size <= 0)
    return;
  if (size > SPI_FLASH_SEC_SIZE)
    size = SPI_FLASH_SEC_SIZE;

  size = (size + 3) & (~3);

  //In case begin() is called a 2nd+ time, don't reallocate if size is the same
  if(_data && size != _size) {
    delete[] _data;
    _data = new uint8_t[size];
  } else if(!_data) {
    _data = new uint8_t[size];
  }

  _size = size;

  noInterrupts();
  spi_flash_read(_sector * SPI_FLASH_SEC_SIZE, reinterpret_cast<uint32_t*>(_data), _size);
  interrupts();

  _dirty = false; //make sure dirty is cleared in case begin() is called 2nd+ time
}``

Semmes that second call destroy data from first call.

SV-Zanshin, 2019-05-22:
The code indicates that the old data is only destroyed and space re-allocated if the size is different. Perhaps then adding 2 parameters to the instantiation call might be the solution :

`INA_Class INA(2056,1024); // instantiate the class with EEPROM size 2Kb and offset 1Kb`

SV-Zanshin, 2019-05-25:
I'm on the road (or, better put, at sea) at the moment and don't have access to an Arduino. Once I get access I'll look into making these changes and will see if that can be implemented.

SV-Zanshin, 2019-09-02:
I've just looked into the open issues - sorry for the delay. I'll see about changing the class instantiation to accommodate this

SV-Zanshin, 2020-03-22:
Sorry for the lengthy delay, this issue slipped between the cracks. Now that I have some time on my hands I will look into it.

SV-Zanshin, 2020-05-18:
Here's an option - I can expose a public variable, "EEPROM_offset", which will be applied to all EEPROM Read & Write commands. The default is "0" but another value can be written to the variable before the begin() method is called. That way a specific number of bytes at the beginning of the EEPROM range can be reserved.

fg89o, 2020-07-11:
What is the state of this issue? I have the same problem with an ESP32

SV-Zanshin, 2020-07-12:
I still don't want to change the instantiation or add ESP32 and ESP8266 specific code, so I think that the last suggestion, exposing a variable "EEPROM_offset" which can be set prior to calling "begin()" which will apply an offset to the first EEPROM address used.

PR #61 (2020-07-12): Development v1.0.12
# Description
Reformatted to standard "Google" c++ using clang-format and implemented enhancements and bug-fixes

Fixes # (issue)
Closes #60  @sages - Fixed possible overflows in computations for getBusMicroAmps() and getBusMicroWatts().
Closes #41 @SV-Zanshin - Added public variable "_EEPROM_offset" to allow reserving EEPROM space.
## Type of change

_Please delete options that are not relevant._

- [X ] Bug fix (non-breaking change which fixes an issue)
- [X] New feature (non-breaking change which adds functionality)
- [X] This change requires a documentation update

# Ho [... cut]
closed 2020-07-12 by SV-Zanshin