Zanduino/INA #42: Initializing INA device does reset RTC DS3231M (RTClib)

Cypher315, 2019-05-23:
Hi !

I have an I²C bus with an INA226 and an RTC DS3231M. I'm using RTClib. When in my code I initialize my INA 
`INA.setI2CSpeed(INA_I2C_STANDARD_MODE);`
		`INA.begin(3, DEFAULT_R_SHUNT ,0);                 //Init INA device`
		`INA.setBusConversion(8500);                           // Maximum conversion time 8.244ms`
		`INA.setShuntConversion(8500);                       // Maximum conversion time 8.244ms`
		`INA.setAveraging(128);                                    //Average 128 readings`
		`INA.setMode(INA_MODE_CONTINUOUS_BOTH);`
It does something weird with my RTC, it resets the minutes and the seconds from the RTC which I don't get at all. What I've read from the source code is that it "scans" the I²C bus from the address 0x40 to 0x80 (my INA device has 0x40 as address) and the RTC address is 0xD0...

I could track down to problem and reproduce it with this pseudo code :
`DateTime dt = RTC.now();`
`print(dt);`
`InitINA()`
`dt = RTC.now()`
`print(dt);`

which output something like this:
`8h43m15s`
`Initializing INA`
`8h0m0s`

Note that this happens only when initializing the chips, when after that I read both Bus Voltage and Bus Current the RTC works perfectly fine.

I have a tight deadline so I wasn't able to find which of the INA method produces this but when I get the time i'll update this post. 

Thanks a lot for this great library, It helped me a lot ! Have a great day,

Anthony

avaldebe, 2019-05-23:
As far as I know the supported INA devices have addresses between 0x40 and 0x4F,
so restringing the [scanning][] makes sense.

@SV-Zanshin what do you think?

[scanning]: https://github.com/SV-Zanshin/INA/blob/master/src/INA.cpp#L241

SV-Zanshin, 2019-05-23:
Yes, the current scan method will reset the DS3231M and @avaldebe suggestion will work and makes sense. The code has been changed at line #241 to restrict scanning to a maximum I2C address of 0x4F

closed 2019-05-23 by SV-Zanshin