Zanduino/INA #43: No way to check ConversionReadyFlag only waitForConversion is possible.

miky2k, 2019-05-28:
To coordinate reading of data from ina226 registers is necessary to read ConversionReadyFlag,
but no function can done without call waitForConversion but that function in not cooperative multithread friendly (in esp8266 is necessary to call yeld or delay function when waiting).

SV-Zanshin, 2019-05-28:
The existing routine can't be changed that way due to backwards compatibility issues with existing code. But it does open up the possibility of creating a conversionState() function which could return the state for either a single device or all devices. Let me think upon this.

SV-Zanshin, 2019-09-02:
I lost this open issue, sorry!  I'm working on adding a new function called "conversionFinished" which returns a true or false for a given device.

SV-Zanshin, 2019-09-02:
I've added the new "conversionFinished()" method.

closed 2019-09-02 by SV-Zanshin