Zanduino/INA #45: Wrong  VBUS Reading

jobitjoseph, 2019-07-21:
The device gives wrong vbus reading . 
for example please see the table below
DMM reading                 INA226 Reading
2.467                                 2.60
4.976                                 5.24
7.64                                   8.05
10.00                                 10.53
       the current reading is correct.

SV-Zanshin, 2019-07-22:
Which INA are you using? Are you running an example program or your own code? Have you tested it with another chip?

The reason I am asking is that the bus voltage is read directly from the INA device and not changed in the library at all, and these readings all look like a similar percentage offset.

jobitjoseph, 2019-07-22:
> Which INA are you using? Are you running an example program or your own code? Have you tested it with another chip?
>
> The reason I am asking is that the bus voltage is read directly from the INA [...]

Hi I'm using INA226 modules, i Have tried The sample program as well as my own. Both gives the same result. And i have tried Multiple module still the same results.

SV-Zanshin, 2019-07-22:
I am away from computers at the moment so cannot test this directly.  Can you do a small program that uses the following functions and displays the values:

getBusRaw(yourDevice);  // This will return that actual bits in the Bus voltage register
getBusMilliVolts(yourDevice); // This will return the raw value multiplied by the device LSB

and the value from your multimeter for the actual voltage at the time.

jobitjoseph, 2019-07-24:
> getBusRaw(yourDevice);
Hi sorry for the late reply, was busy with work. please find the reading below 
DMM	   INA		raw  
2.522	2.6480		2119      
3.309	3.4760		2781
5.023	5.2780		4223
7.49 	7.8730		6299
10.07	10.5900		8472
12.46	13.1120		10490
14.96	15.7280		12582
17.52	18.4230		14739
20.07	21.1050		16884

SV-Zanshin, 2019-07-24:
If you divide INA by RAW you will always get exactly 0.00125. This is what internal LSB value is and therefore the library is indeed returning the correct value for what is being measured.

Thus I believe it is not a bug in the library but a problem in the hardware. At first blush I can think of the following possibilities;

1. Invalid INA Device. Have you tried using another INA chip?
2. Incorrect measuring point for voltage. Is the DMM measuring at the same junction that the INA device is measuring?
3. Could the DMM be incorrect?

jobitjoseph, 2019-07-25:
> If you divide INA by RAW you will always get exactly 0.00125. This is what internal LSB value is and therefore the library is indeed returning the correct value for what is being measured.
>
> Thus I [...]

I have tried 3 different Ina modules.
Yes I'm measuring at the same junction.
I have tried 4 multimeters and 2 panel meters. Only difference is the panel meter reading is around 2mV off.

I don't know what is going on. The current reading is spot on. But vbus reading is giving this 5% off reading

SV-Zanshin, 2019-07-25:
Even though I can't see a problem in the code, could you try using another INA library to see if that also has the same error?

SV-Zanshin, 2019-07-27:
Any luck with another library? It would be nice to know that the error is reproducible regardless of which INA library is used, or that the error only occurs with my library.

jobitjoseph, 2019-07-27:
> Any luck with another library? It would be nice to know that the error is reproducible regardless of which INA library is used, or that the error only occurs with my library.

Hi, currently I'm out of town. I will get back to you as soon as I return. Thank you.

jobitjoseph, 2019-08-02:
> Any luck with another library? It would be nice to know that the error is reproducible regardless of which INA library is used, or that the error only occurs with my library.

Hi sorry for the long hold. I have tried other libraries as you suggested and all gives the same ,5% error. I can't pinpoint what's the actual issue. The current reading is extremely accurate , we can assume the shunt voltage too.but when it comes to vbus don't know what is really happening

SV-Zanshin, 2019-08-02:
OK, at least we've ruled out Software issues.  

You stated that you've tried different INA226 modules so the module itself shouldn't be a defective one. Could you measure the voltage directly from the 2 Pins on the INA226 to make 100% certain that you are measuring the same Thing that the INA is?

jobitjoseph, 2019-08-03:
> OK, at least we've ruled out Software issues.
> 
> You stated that you've tried different INA226 modules so the module itself shouldn't be a defective one. Could you measure the voltage directly from the 2 Pins on the INA226 to make 100% certain that you are measuring the same Thing that the INA is?

Hi I have already done that. If we change the value of LSB to 1.19 it gives accurate vbus reading. I know the LSB is 1.25 by default, but in this situation it's not. May be these chips at counterfeit ( it's the only explanation but can think of now).thanks for your help buy the way. Have a great day.

SV-Zanshin, 2019-08-04:
I'll close this out for now as it really doesn't seem to be a library issue and I hope that you get it sorted out. It could have been a bad batch that shouldn't have been sold.

closed 2019-08-04 by SV-Zanshin