Zanduino/INA #62: Hardcode EEPROM size

fg89o, 2020-07-12:
I think there should be a posibility to set EEPROM size in begin() method. Before your last merged I did a test with an EEPROM_offset and works correctly but changing the line 

`EEPROM.begin(512).`

For example, If I need 400 bytes for my main project  and I want another 512 bytes for INA library the begin method should set EEPROM to 912. Right?. You can expose antoher variable to set EEPROM size and initializate to a default value to have backwards compatibility.

In the other hand, you calculate max_devices including the EEPROM_offset variable ,If the library is not allowed to use the offset, the maximum devices should be:

`maxDevices = (EEPROM_size - _EEPROM_offset  ) / sizeof(inaEE);`

instead of

`maxDevices = (_EEPROM_offset +   EEPROM_size) / sizeof(inaEE);`

I did all the tests with an ESP32. Hope something of that help you to improve the library,

SV-Zanshin, 2020-07-12:
:thumbsup: I missed that, good catch and I've added the correct offset math to the code.

fg89o, 2020-07-12:
Are you going to create the EEPROM size variable too?.

SV-Zanshin, 2020-07-12:
I wanted to fix the actual bug first, before the new release was published.

I still need to think about changing the begin() method, which is why this issue is still open.

SV-Zanshin, 2020-07-12:
What I will do is add another public uint16_t variable "_EEPROM_size" which is only valid for the ESP32 and ESP8266 platform. It defaults to 512 but can be changed before the begin() is called.

fg89o, 2020-07-12:
Perfect! Thank you for your support.

SV-Zanshin, 2020-07-12:
OK, that should work!

closed 2020-07-12 by SV-Zanshin