Zanduino/INA #64: begin does not set values on subsequent calls

johntaves, 2020-08-08:
I call begin with 2 parameters to discover the devices. I passed the maxamps and microohms arguments and those are applied to all devices. I then call begin with 3 parameters to set the 2 values on device 1 to be different than what I set for device 0, but those values do not take.

I believe that the fix is:

  if (_DeviceCount == 0)  // Enumerate all devices on first call
  {
...
  } else {
    readInafromEEPROM(deviceNumber);  // Load EEPROM to ina structure
    inaEE.maxBusAmps = maxBusAmps > 1022 ? 1022 : maxBusAmps;  // Clamp to maximum of 1022A
    inaEE.microOhmR  = microOhmR;
    ina = inaEE;
    initDevice(deviceNumber);
  }                         // of if-then-else first call

This change seems to work for me. I did not spend the time to be confident that I understand the code.

However, this seems like a poor design. Why not have begin() take no arguments and return the number of devices and have a separate setXXX to set the 2 values for each device?

johntaves, 2020-08-09:
I am using an ESP8266. It seems to me that the EEPROM will be used. Am I confused?

If it is being used, what prevents another library from using the same address space in the EEPROM?

SV-Zanshin, 2020-08-10:
I will look into the first issue, but changing the functionality of the begin() call in the manner you suggest is not possible, as it would break existing code.

To address the second question - there is nothing to prevent another library or the programmer from using the same address space. The library allows the EEPROM to be relocated, but there is no mechanism to prevent any other code from accessing the same locations.

johntaves, 2020-08-10:
I don't see the point of using the EEPROM. I just need to store the maxamps and ohms for my devices and do so whenever the user changes them. The other values that the INA lib uses are all derived at run time and thus don't need to be EEPROMed.

I modified my instance to not use the EEPROM, instead using the array. I also sized the array to my max number of devices. 

I suggest that there be a version of begin that takes a pointer to a array of inaEE and max size parameter, and returns the # of devices. And add a setXXX function to set each devices' maxamps and ohms.

These changes will not affect any existing code that depends on the begin(x,y) version. Note that nobody is using the begin(x,y,z) version, because it does not work.

johntaves, 2020-08-11:
Also the code that mods _DeviceCount back to 0 when more devices are found than there is space makes no sense (See: // start again at 0 if overflow). Change that to a simple _DeviceCount++; and it will work fine.

fg89o, 2020-08-14:
would be a good option to release a 2.0 version with optional parameter to use EEPROM or not. I could test it with ESP32. I got a little confused about the EEPROM at the beginning too

SV-Zanshin, 2020-11-30:
Sorry for taking so long to address this issue.  I will look into making the use of EEPROM for storing the memory structure optional - but I'll open up a separate issue for that

SV-Zanshin, 2020-11-30:
Fixed as of v1.0.13

closed 2020-11-30 by SV-Zanshin