Zanduino/INA #66: Possible error with INA226

SV-Zanshin, 2020-08-30:
Taras sent in the following:

Hi there. I used your library for INA226. Found an error. In the example DisplayReadings.ino
power is not calculated correctly. I add a screen.
V = 8,843 V
A = 35.736 mA
V * A = 8.843 * 35.736 = 316mW
, and in the example it turns out 253 mW.
Could this be a library error? Maybe it's my fault?

--
[image]

SV-Zanshin, 2020-08-30:
I am looking at the library and can't locate an error. The "getBusMicroWatts()" function will read the next computed value, so if the current changes very quickly then you might get a value that doesn't correspond to the BusmA * BusVoltage. Could this be the case for you? Is the problem reproduceable?

Samodelkin-YouTube, 2020-09-20:
Today I tried to use the library  "INA226" ,everything works well there.
   A 0.1348      V 19.120     W_ina 2.5788      W=V*A   2.5774      % 100.05

When I use the "INA" library, the power is incorrect, the error is 80%

  A 0.1348         V 19.118     W_ina 2.0618     W=V*A 2.5771        % 80.01

SV-Zanshin, 2020-09-20:
Since the Amps and Volts are the same in both cases, the problem lies in the computation of the Watts.  Are you initializing both libraries with the same begin() statement?

Samodelkin-YouTube, 2020-09-20:
"INA226.begin " and  "INA.begin"

SV-Zanshin, 2020-09-21:
I meant the parameters to the begin() - are they identical, especially the values for expected maximum amps

Samodelkin-YouTube, 2020-09-21:
all the same

SV-Zanshin, 2020-09-27:
How comfortable are you to make some small changes to your version of the library?  I would like you to change the library by adding the following after line 634 "(int64_t)readWord(INA_POWER_REGISTER, ina.address) * (int64_t)ina.power_LSB / (int64_t)1000;"

The lines are:

`Serial.print("power register is ");
Serial.println(readWord(INA_POWER_REGISTER, ina.address));
Serial.print("power_LSB is ");
Serial.println(ina.power_LSB);`

Gallium70, 2020-10-24:
According to the INA226 datasheet, page 16, for INA 226, the power LSB should be 25 * Current_LSB:
[image]

However, it's set to 20x in INA.cpp, line 27:
[image]

And 20/25 is exactly 80%.

SV-Zanshin, 2020-10-25:
Good sleuthing - I am re-installing my PC this weekend (Thank You, Mr. Gates) but once all the software is installed I'll check this out.

Samodelkin-YouTube, 2020-10-25:
> Согласно [таблице данных INA226](https://www.ti.com.cn/lit/ds/symlink/ina226.pdf) , стр. 16, для INA 226 младший разряд мощности должен быть равен 25 * Current_LSB:
.
Why is this necessary? 
P.S. Installed the library INA226 ,everything works fine.

Gallium70, 2020-10-25:
> > Согласно [таблице данных INA226](https://www.ti.com.cn/lit/ds/symlink/ina226.pdf) , стр. 16, для INA 226 младший разряд мощности должен быть равен 25 * Current_LSB:
> > .
> > Why is this necessary?
> > P.S. Installed the library INA226 ,everything works fine.

As the datasheet explains,

> This ratio is internally programmed to ensure that the scaling of the power calculation is within an acceptable range.

Below is my theory about this. The current and bus voltage are both stored in 16-bit registers. If you directy multiply them together, you'll need a 32-bit register to store the result. To put it into a 16-bit register, scaling is necessary(of course there's a loss in precision), and the scale factor should be somewhere near 2^(32-16)=65536, intuitively.
TI chooses a scale factor of 20000, since the LSB for the bus voltage is fixed at 1.25mV/bit (equivalent to 800x scaling), then the power LSB should be 20000/800=25x current LSB.
As for why it's exactly 20000 (not some other number), maybe only TI knows.

SV-Zanshin, 2020-10-25:
I see the differences in the DataSheets, but before I commit the changes, I will need to hook up a INA219 and INA226 and test the changes.  Still downloading MS-Studio, VisualMicro and need to install doxygen, clang-format, gitKraken, etc.

 

-Arnd.

SV-Zanshin, 2020-10-26:
First test looks good :)

SV-Zanshin, 2020-10-26:
Fixed in Development version 1.0.13, which should be promoted to a new master version in the next couple of weeks.

closed 2020-10-26 by SV-Zanshin