Zanduino/INA #67: {"Trying to understand implementation of INA3221"=>""}\

andyrbarrow, 2020-09-06:
I've implemented example programs using the INA3221. They seem to work, with one part I don't understand: Currently, when a single INA3221 is connected (using ESP32), it is discovered as 3 devices on address 64. In the DisplayReadings example program, it lists three devices, and it appears that all three devices have the same readings.

How does one differentiate between the three channels of the INA3221?

andyrbarrow, 2020-09-06:
Okay, my mistake

andyrbarrow, 2020-09-06:
My mistake. I was trying to use one of those breakout boards that has the grounds from all three channels common. It wasn't until I separated them that I was able to see independent voltages on all the channel.

closed 2020-09-06 by andyrbarrow

SV-Zanshin, 2020-09-07:
Hello Andy - glad you got that sorted out. I had the same board and the same problem for the INA3221 and was pretty disappointed in how they put that board together.

andyrbarrow, 2020-09-07:
Yes. The only saving grace is that I ordered a single unit  from Amazon and
they sent me a bag of 29! So, I have lots to play with.

I have two battery banks to monitor, voltage and current. My plan at this
point is to implement two of these INA3221 boards with the traces cut and
the shunt resistors removed. Each one will measure separate banks -
low-side current with an external 75mV shunt on one channel and battery
voltage on a second channel. Then I avoid the work of moving shunts,
remounting, rewiring, etc. If anyone asks about this in the future,
there is a guy on Tindie who sells INA3221 breakout boards that are wired
correctly. If I didn't have so many of the other ones, I'd definitely be
buying a few from him.

Thanks for an excellent library and impeccable documentation (both in code
and in the external docs). I'll let you know how it goes.
*Andy Barrow*

*email: *andy@sailor.nu
*tel (mx):* +52 322 104 2189
*tel (us):* +1 925 270 0473
*Skype:* abarrow

ᐧ

SV-Zanshin, 2020-09-07:
Hello Andy,

 

How big are your battery banks? I’ve got two really big banks, so I have to get really big shunts (the house bank is 800Ah @ 24V – so 1600Ah @ 12V, the smaller bank is only 200Ah @12V and then there’s the starter battery for the engine @12V). Since I charge at around 28.8 most of the INA devices won’t carry enough voltage for me.

 

-Arnd.

 

*email: *andy@sailor.nu <mailto:*andy@sailor.nu>

andyrbarrow, 2020-09-07:
I'm using 200A shunts, mainly because the last monitor that I installed
came with one, so I only had to buy one more. These are fairly cheap shunts
that I got on Banggood for $6.50 USD (I think that's right). I 3D printed a
holder for them, so I could tie the grounds together.

I have a 12V 400Ah bank and a 12V 100Ah starting battery. I probably could
get away with smaller shunts, but at least with these I can still have
them in the motor starting circuit.

I suppose you could put in a voltage divider to bring the voltage down to
something that the INA would be able to monitor.
*Andy Barrow*

*email: *andy@sailor.nu
*tel (mx):* +52 322 104 2189
*tel (us):* +1 925 270 0473
*Skype:* abarrow

ᐧ

yesyesuk, 2020-09-19:
> ... there is a guy on Tindie who sells INA3221 breakout boards that are wired correctly. 

I'm that guy. ;-)

andyrbarrow, 2020-09-19:
Cool! In my case, I removed the shunt resistors completely, as I wanted to
use my 200 A shunts. So, on the the "bad" boards, two trace cuts to
separate grounds (as your boards do), and pull the resistors off. I know
you have a couple of versions of that board - it might be worth offering
them without shunt resistors.

SV-Zanshin, 2020-09-20:
Do you have a URL for the boards?

yesyesuk, 2020-09-20:
Yes, I do. I just didn't want to barge in here advertising my boards. ;)

0.01 Ohm version:
https://www.tindie.com/products/yesyes/improved-ina3221-breakout-3-power-supplies-001ohm/

0.1 Ohm version:
https://www.tindie.com/products/yesyes/improved-ina3221-breakout-3-power-supplies-01ohm/

I'll consider having some made without shunt resistor with the next batch.

SV-Zanshin, 2020-09-20:
Awesome - thanks for the link. Once I get my things sorted out here I think I might order some!

yesyesuk, 2021-01-13:
>it might be worth offering them without shunt resistors.

I now have a model without shunt resistors available. ;-)
https://www.tindie.com/products/yesyes/improved-ina3221-breakout-3-power-supplies-no-res/

andyrbarrow, 2021-01-13:
Thank you! I just ordered 4!
*Andy Barrow*

*email: *andy@sailor.nu
*tel (mx):* +52 322 104 2189
*tel (us):* +1 925 270 0473
*Skype:* abarrow

SV-Zanshin, 2021-01-13:
@yesyesuk And I just ordered 2.  I only need one, but I adhere to the maxim "one is none" and believe that breadboards tend to release magic smoke once in a while.
@andyrbarrow  I obfuscated your details in the previous post - it is better to not broadcast too much information for those screen-scrapers to harvest

yesyesuk, 2021-01-13:
@andyrbarrow @SV-Zanshin 
That's odd, I did not receive any orders.

SV-Zanshin, 2021-01-13:
I ordered at 06:45 this morning and have received confirmation “Processing Order #240740 | Tindie”

yesyesuk, 2021-01-13:
hmmm, maybe it takes a while for them to process the order. I've never been told outside Tindie when someone ordered. So I don't know what the normal delay is.

I will keep an eye on it.

andyrbarrow, 2021-01-13:
Thanks for removing my signature, sometimes I forget!

I did get a confirmation from Tindie for the 4 that I ordered as well. Even
though I paid using PayPal, there was a comment in the order confirmation
about waiting a few days for the payment provider to confirm the payment.

One minor suggestion on the description of the part - in my case I'm using
it with an external high current shunt. While that sort of usage is clearly
implied in your description, it might help you sell units if you are more
explicit about that usage mode.

Heck, if you are feeling like expanding your line, why not sell a
i2c-enabled high current shunt?
ᐧ

yesyesuk, 2021-01-13:
@andyrbarrow 
Great idea regarding mentioning external shunts. I have added that to the Tindie product description (will take a few minutes to update publicly) and to the longer description on my website.

I have contacted Tindie support regarding these 2 orders. Could you please tell me your order number? I can add this to the ticket so they can have a look.

andyrbarrow, 2021-01-13:
240730

ᐧ

yesyesuk, 2021-01-13:
OK, all good. Tindie support said that these orders were caught by their filters and needed to be approved manually (by them). I guess because it was just a few hours after the new product went live.
I have received the orders now and will ship tomorrow (sorry, too late for today).