Zanduino/INA #69: Could you please give a full example for the code to run several INA modules?

rin67630, 2020-11-23:
Call me stupid, but i cannot get the correct way to get the measurements out of two ina226 located e.g at 0x40 and at 0x41.
Where am I supposed to enter the module address?

Could you provide an example? That would make it easier for everybody.

rin67630, 2020-11-23:
Additionally, you write that you are using EEPROM to store the configuration data
Where do you store this data? How large is it? Can a user influence the storage location?
We need the info to prevent conflicts with other usages (from user code or libraries) of the EEPROM.

SV-Zanshin, 2020-11-24:
The wiki page for [_EEPROM_offset](https://github.com/SV-Zanshin/INA/wiki/_EEPROM_offset) describes that although the default offset is at the beginning of the EEPROM (offset 0), it can be shifted backwards by changing this value. Note that this needs to be done before the begin() call.

While the calls to the functions optionally use a "deviceNumber", the corresponding I2C address for each device found can be determined by a call to [getDeviceAddress()](https://github.com/SV-Zanshin/INA/wiki/getDeviceAddress()). But the I2C bus search function is sequential, so a lower I2C address will get a lower device number.

SV-Zanshin, 2020-11-24:
Let me search through my box of components and sort out several INAs and go about making an example script.

SV-Zanshin, 2020-11-25:
I just realized that the example program [DisplayReadings.ino](https://github.com/SV-Zanshin/INA/wiki/DisplayReadings.ino) supports multiple devices. Is that not what you were asking for?

SV-Zanshin, 2020-11-28:
I'll close this issue soon if I don't hear back, as I think that there is already an acceptable example available.

Gallium70, 2020-11-28:
0x40=device 0
0x41=device 1
0x45=device 2

SV-Zanshin, 2020-11-29:
@rin67630  -  I haven't heard back from you on whether or not you consider the example program acceptable and am therefore going to closet his issue but don't hesitate to re-open it if you think otherwise.

closed 2020-11-29 by SV-Zanshin