Zanduino/INA #70: How is power calculated?

markusoestreicher, 2020-11-27:
I have connected an INA226 to a 50A/75mV or 0,0015 Ohm shunt and uploaded the DisplayReadings example to my NodeMCU. 
Shunt voltage is read and the 200 mA current is calculated properly and verified by a Multimeter.

However, the Power output looks wrong. 5V*0,2A should be 1A, but I get readings around 0,8W all the time. Any idea?

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  64 INA226  5.0410V    0.3020mV  201.4220mA  793.4810mW

Thanks!

SV-Zanshin, 2020-11-27:
What does your [begin()](https://github.com/SV-Zanshin/INA/wiki/begin()) call in the program look like? This is where you tell the INA what your expected maximum amperages and what the shunt resistance is.  In this case it should read "INA.begin(50,1500)" for 50A maximum and 1500 microohms.  The power is computed internally by the INA using these 2 values so if power readings are off it is almost certainly due to incorrect settings here.

Gallium70, 2020-11-27:
I guess it's the same issue with #66 . You may try the 1.0.13 version branch and see if it's solved.

SV-Zanshin, 2020-11-27:
@Gallium70 - I had forgotten that change
@markusoestreicher  - which library version are you using?

markusoestreicher, 2020-11-28:
I checked my historic data and the power measurement was correct at some point.

I noticed I am already on 1.0.13, so I made a simple setup with the DisplayReadings example and a 0,2A constant load.

Not touching the setup, I just downgraded my library to 1.0.9 and re-compiled each time.

This is how it looks on 1.0.13:

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  65 INA226  4.9910V    0.3120mV  207.5250mA  854.5180mW

Then I downgraded to 1.0.9 and the measurements are correct:

1.0.9:

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  65 INA226  4.9930V    0.3100mV  205.9990mA 1029.9990mW

1.0.10:

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  65 INA226  4.9930V    0.3120mV  207.5250mA  854.5180mW

and back again to 1.0.9 for verification:

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  65 INA226  4.9920V    0.3120mV  207.5250mA 1068.1470mW

The change must be in 1.0.9 -> 1.0.10

Then I analyzed the changes between 1.0.9 and 1.0.10 and found the regression.
Fixing issue 54, you moved power_LSB above the switch but modified it from 25 to 20 for the INA226.

If I take 1.0.13 and change INA.cpp line 27 like this:
power_LSB     = (uint32_t)**25** * current_LSB;                 // Fixed multiplier per device
I get proper readings again:

Nr Adr Type   Bus      Shunt       Bus         Bus
== === ====== ======== =========== =========== ===========
 1  65 INA226  4.9900V    0.3120mV  207.5250mA 1068.1470mW

However, that would be wrong for other variants, so I suggest moving the power_LSB back into the switch/case.

markusoestreicher, 2020-11-28:
I missed the reference to issue #66. So, this is the same issue.

SV-Zanshin, 2020-11-29:
I gather from your post that I can close this issue. I will make a couple more changes to the library and then publish a new release that incorporates the fix for issue #66 in the next couple of days

closed 2020-11-29 by SV-Zanshin