Zanduino/INA #72: Allow an option for the library to not use EEPROM storage for INA-Structure information

SV-Zanshin, 2020-11-30:
**Describe what the feature request is and if it solves an issue or adds functionality**
Each INA device consumes 20 bytes of RAM. Since this library allows many INA devices to be connected at the same time, this has the potential of using up a significant amount of an Arduino's scant memory. Hence the use of EEPROM memory to store this information - as the EEPROM storage is seldom used this makes good use of otherwise wasted RAM.
But there have been questions as to whether it might be possible to _not_ use EEPROM and to store this information in  memory after all

**Describe the solution you'd like**
Add a switch to the library to disable EEPROM (default behaviour remains as currently implemented with EEPROM storage used), and when EEPROM use has been disabled store the information as a dynamic array in RAM.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

rin67630, 2020-12-01:
+1, Backing up this request.

SV-Zanshin, 2020-12-01:
I have opted to change the class constructor to add a parameter there which will let the library determine whether to use dynamically allocated memory or EEPROM to store INA structure information.  While each INA structure element is only 20 bytes, this can add up quickly when several INA devices are implemented - particularly when working with an UNO rather than the newer and bigger microprocessing systems.
If the class is instantiated using a numeric parameter greater than 0, the library will allocate memory and not use EEPROM. So, a definition of "INA_Class INA(5);"  will disable EEPROM use and reserve a 5-element array internally (currently 100 bytes). Since this is done during class instantiation no error checking is done. If more memory is reserved than is available on the heap with a "new" call then no error is returned, but the program will abort when accessing a device. Similarly, if there are 7 devices available then the program will also abort when attempting to access the 6th device with the initializer given above.

SV-Zanshin, 2020-12-02:
Created Version 1.0.15 for this fix and updated documentation

closed 2020-12-02 by SV-Zanshin