Zanduino/INA #73: INA3221 is supported, how to configure?

rin67630, 2020-12-01:
Hi Zanshin,
let me first apologize for not having confirmed the previous issue. Yes i did understand now how it works.
My next question arises:
I have ordered some INA3221 that you are supporting. I just wonder how to configure the channel.
That way?
  ina1_shunt1     = INA.getShuntMicroVolts(0,0);
  ina1_shunt2     = INA.getShuntMicroVolts(0,1);
  ina1_shunt3     = INA.getShuntMicroVolts(0,2);
  ina2_shunt1     = INA.getShuntMicroVolts(1,0);
Thank you for clarifying

rin67630, 2020-12-01:
I have seen that #andybarrow asked the same question:
"How does one differentiate between the three channels of the INA3221?"
but I did not find the answer in the dialog following.

SV-Zanshin, 2020-12-01:
The INA3221 shows up as 3 distinct devices. The "getShuntMicrovolts()" function only takes one parameter, so you would call it as
```
   uint16_t in3211_1 = INA.getShuntMicroVolts(0);  
   uint16_t in3211_2 = INA.getShuntMicroVolts(1);  
   uint16_t in3211_2 = INA.getShuntMicroVolts(2);  
```

SV-Zanshin, 2020-12-01:
Where die @andybarrow ask the question?

rin67630, 2020-12-01:
Thank you! 
@andybarrow asked in closed issue #67.

Since you both seem to be interested in building BMS with INA chips, I am working on a networked battery reporting system based on INA chips, ESP8266 and thinger.io as dasboard server.
I mainly aim to build a software MPPT around plain buck converters, but the main code could be used for a BMS as well.
The main feature is currently to be able to forward the battery information over the network to another ESP, so one could build a BMS for any voltage with an ESP per 12V block, overcoming the 26V limitation of the INA.
https://github.com/rin67630/Solar-Booster
Stay tuned...

closed 2020-12-01 by rin67630

rin67630, 2020-12-01:
here an example of my solar-booster dashboard:
[image]

SV-Zanshin, 2020-12-01:
Very nice!

My battery bank is 24V which means it can hit 28.8V while charging so the standard breakout boards for the INA219 won't work for me.  I want to monitor my solar panels separately, so the INA3221 is perfect for me so that I can save on components.

rin67630, 2020-12-01:
INA226s could do (36V), but are limited to 24V nominal.

I suggest to use a network of several ESP8266s (one for each block of 12V + one to centralize the values) equiped with INA3221 to monitor each cell, to make a scalable solution, open ended to every voltage.
You have more devices (an ESP8266 costs nothing), but far less cables.