Zanduino/INA #75: Settings for Shunt_Micro_Ohms and Max_Amps

JCHPAUL, 2021-01-26:
Hi, thanks for this code it is a big help. In the process of getting a number of INA devices (INA219, INA226, INA3221) working recently, I came across some issues with regards to the setting of Max_Amps when different values of Shunt_Micro_Ohms are selected. Hence I have complied the attached PDF "cheat sheet" that makes it easier to select valid settings for Max_Amps. 

[Setting of Max Expected Amps in INA2xx.pdf](https://github.com/Zanduino/INA/files/5873252/Setting.of.Max.Expected.Amps.in.INA2xx.pdf)

SV-Zanshin, 2021-01-26:
Let me see if I can integrate the contents of the PDF to the site description.

SV-Zanshin, 2021-09-29:
I've added your document - many thanks!

closed 2021-09-29 by SV-Zanshin