Zanduino/INA #77: INA228 Support

Agilitytime, 2021-03-09:
Hi Zanduino,

thank you very mutch for the Libary.
Is it Possible to Integrate the INA228 ?

here are the Specs 85-V, 20-bit, ultra-precise, I2C output current/voltage/power/energy/charge monitor with alert.
https://www.ti.com/product/INA228

The Cool think are the energy/charge Monitor and 20 Bi resoulution.

Best,
Basti

SV-Zanshin, 2021-03-09:
Those new INA228 and INA229 look nice! I might be interested in them for myself for a solar power charging system.  Unfortunately, these are not I2C but SPI systems, which would mean making some significant changes/additions to the INA library. That doesn't mean that it isn't possible, just that it isn't just a matter of adding the new specifications into the existing code.

At the moment neither Mouser nor Digikey have any in stock, otherwise I would have ordered some already :)

SV-Zanshin, 2021-03-10:
Ahh... I downloaded both datasheets, but only looked at the INA229 one.  I'll see if I can get a INA228 ordered (and soldered) and should be able to add it to the library. I won't start with any coding until I get the hardware, though - so it might take a bit.

SV-Zanshin, 2021-03-25:
I've ordered a INA228 from Digi-Key and will see about soldering together a test board and adding this device to the library.
The INA237 is still not in stock.

SV-Zanshin, 2021-04-05:
The INA228 has arrived. I thought that the package was a 10-TSSOP and I have a breakout adapter to DIP for that, but looking at the chips they seems awfully small and I re-checked the datasheets to find that the package is a 10-VSSOP and I have no adapters for that. A quick look at digikey didn't show me any adapters, so I might have to draw up a schematic and order some special PCBs to get this done; as there are still no breakout boards out there.

Agilitytime, 2021-04-05:
Hello Arnd,

thank You for your Support!!!
Can i "Help" you with a donation ? 

Thanks,
Sebastian

SV-Zanshin, 2021-04-05:
Hello Sebastian (that sounds like a German name), 

Thanks for the kind offer!  I will download Fritzing again and try to design a board that I can get cheaply made. I have a toaster reflow oven I made a long time ago that needs to see some use :)

I just am not proficient with Fritzing or similar CAD products so this might take a while.

aXe-88, 2021-05-02:
Hi,

I am looking into pulling myself and contribute. Before you design a Board just use a INA226 based one and exchange for the 228. Same pinout, same package.

If you want i'll send you one.

Br, Alex

SV-Zanshin, 2021-05-03:
Hello Alex,

 

That would be great. If I had a prototype I could change the library and test the changes.  I haven’t designed a board yet, I got caught up in making an Arduino controlled coffee roaster (a popcorn popper, solid-state relay for the heater PWM, MOSFET for the fan PWM, two thermistors, LCD display, two rotary encoders and lots of software 😊 )

 

-Arnd.

daned33, 2021-05-09:
@SV-Zanshin

I've ordered these chips, and have breakout boards arriving next week. Will be willing to test!

If you send me your mailing address I can mail you some adapters.

SV-Zanshin, 2021-05-10:
Hello Daniel - Alex has been kind enough to offer sending me an adapter with the INA228; although I'm not sure when. But if that doesn't work out I'll get back to you and your generous offer.

daned33, 2021-05-10:
Okay good to hear, send a message if you still would like some boards.

I'd be willing to take a shot at implementing the 228 once my boards and chips arrive.

SV-Zanshin, 2021-05-10:
Perhaps you could test the library version with INA228 support once I get that done, and you can make sure that I didn’t miss anything.

-Arnd.

daned33, 2021-05-10:
That works for me!

SV-Zanshin, 2021-05-12:
@aXe-88  - I just checked my mailbox and have received your board with the INA-228! Many thanks for that; and since tomorrow is a national holiday and the weather is supposed to be cold and rainy I think I'll get started with putting this into a circuit and developing the library extension.

THANKS!

SV-Zanshin, 2021-05-13:
I started working on this today and have discovered a basic issue with the library which I can't quite figure out (the "inaDat" class keeps on getting instantiated and I can't see why). So this might take a bit longer than I originally thought to get implemented.
[image]

SV-Zanshin, 2021-05-16:
Update: This might take a bit longer than I originally thought. The INA228 is a different "beast" from the others and I'd originally thought I'd only need to change some of the internal values to add this chip; but the accuracy is 20bit rather than 16bit so I have been changing internal variables to 32bit and am debating taking the plunge and changing the library to use floating point or perhaps even double.

magicsmoke, 2021-06-22:
Hello Arnd.

I also have the INA228 now to test with once you get a initial release out :)

yahaggach, 2021-09-06:
Hello SV-Zanshin,

Could you please tell me where you got the breakout board for the INA228 is it a small one you designed yourself or readily available somewhere ?

Thanks !

SV-Zanshin, 2021-09-06:
Unfortunately I didn't find a commercial source, the pinout is the same as INA226 and Alex sent me a prototype to use.

yahaggach, 2021-09-06:
Thanks a lot. I'll quickly grab a breakout board for the INA226 and replace the chip with a 228. If you don't mind, in terms of accuracy and resolution, what INA chip do you recommend for voltages <36V and currents <5A. My main need is high resolution and accuracy.

SV-Zanshin, 2021-09-06:
I'm not in place with fast internet at the moment and don't have the data sheets handy, but I think that the Texas Instruments site has a chart of the various devices including the voltage ranges and accuracy. Or perhaps using DigiKey to list the INA2xx devices and then sorting by accuracy/gain?

yahaggach, 2021-09-06:
Fair enough. Thanks for the help !

majianjia, 2021-09-12:
Hi @SV-Zanshin 
Is the current lib supported 228?

SV-Zanshin, 2021-09-13:
@majianjia  - not yet. I am currently away from my home but plan on returning soon and will then update the library to support the INA228 functionality correctly.

majianjia, 2021-09-13:
@SV-Zanshin Very well, I am trying to implement it but facing the same 20bit registers problem, looking forward to seeing your implementation. Thanks.

varun-skies, 2021-09-24:
Hy SV-Zanshin

any update on the library to include INA228. Have you tested the library ?? Please release the updated library soon.

SV-Zanshin, 2021-09-29:
I'm really sorry for the long delay - I'm currently in lockdown far away from my home computer, the INA228 and my testing hardware so I haven't been able to do any work.  I thought I would have been able to return 2 months ago, but that return journey is proving difficult.  I will be back on November 1st but I think I can find some time soon to program around a bit - but I won't have a hardware test bench with an INA228 available and since that chip is significantly different from the others I don't think that it is is going to work out in the short term.

riscyd, 2021-10-04:
> I'm really sorry for the long delay - I'm currently in lockdown far away from my home computer, the INA228 and my testing hardware so I haven't been able to do any work. I thought I would have been [...]

I have one bult into my hardware, connected to a QT PY module. I could offer remote access if you would like to test some libraries. I have some basic readouts working.
BTW, did you see that TI offers a library builder to download? It is just not arduino compatible as is.

sebashb, 2021-11-02:
Hi @SV-Zanshin, we have a INA228 test board from Texas, we may help you test the library. We are also very interested in suing the INA228. We have already adapted another library that kinda works but sometimes we get wrong current values.

majianjia, 2021-11-02:
@sebashb  Could you also recommend the lib support INA228 for me? Thanks. I am currently using my own code, not very ideal.

sebashb, 2021-11-02:
There is no lib, we adapted one lib that was made for the INA226 to make it work with the 228. But it’s nos really a finished product. Sometimes it works sometimes it doesn’t.

SV-Zanshin, 2021-11-05:
I'm returning home next week and my first Arduino project is going to be finishing the INA228 support for this library. I really want to use the chip and it seems that there is a lot of interest out there for a good library, which I hope to be able to supply.

rapzak, 2021-12-03:
Hi, is there any progress on the INA228, i have a board with 3x INA229 i like to get up running - maybe change this to SPI, so is there some work in progress i could see into to add SPI? :)

jhardesty200, 2021-12-07:
Also pretty interested if there is any progress on the I2C version? =>

SV-Zanshin, 2021-12-07:
I've returned back home a week ago and will be setting up my INA228 test rig this weekend and hope to make some progress on this - I'm excited to get this addition done.

jhardesty200, 2021-12-07:
I am not much help when it comes to software but I'm good with hardware! Let me know if I can be of any help with the project. I am working on a relay charge controller utilizing the INA228 and it would be so much cleaner to have a library XD - good luck and keep me posted!

Robocraze, 2021-12-17:
Hi, We were working with this on our side as well. We have come up with a library with some basic functionality if this helps. You can check it here: https://github.com/Witty-Fox/DataCollector

SV-Zanshin, 2021-12-17:
@Robocraze  - awesome, many thanks! I just took a quick look and, while I will continue with updating my library, I think I can use some of your work in your library to assist in that process. And I can always run both libraries to compare outputs to make sure that my library works correctly.  The main reason that I'm continuing on my side is that I already support other INAs and I also won't use floating point on Arduinos (just a person design decision). Thanks again!

sebashb, 2021-12-17:
Attached you can also find the Library that we have developed, I don't remember exactly in which one we based ours but it is not always giving the correct current values and we can't find why. Nonetheless we think that the rest of the functions, namely the configuration part is correctly implemented since it was debuged with the help of TI Staff.

[INA228.zip](https://github.com/Zanduino/INA/files/7737993/INA228.zip)

riscyd, 2021-12-19:
I have written my own arduino code to write and read the most important registers. Works quite well. 
However I noticed that the sensor is quite noisy, not to say super noisy. Vcc etc. is proper. 
I experience this even if nothing is connected amd pwm is off. With PWM and in operation it's a bit worse as expected. 
I see this in a range of +/- 10-15 bits. With pwm on a bit higher. 

For my application in a MPPT it seems not really usable. Except with very high averaging and sample times which then gives no satisfactionary results in terms of tracking speed. 
The Microchip PAC 193x series seems here much more suitable and WAY less noisy.
I tried without input filter, with 2x 22R and 0.1uF and up to 2x 100R and 1uF ceramic cap. The external filter to my surprise did not help much.
The tracks are proper layouted and short. 

I'd like to hear whether people here made similar observations?

SV-Zanshin, 2021-12-20:
I reserved Sunday for working on the INA228 and didn't get very far at all. I couldn't find the breakout that @aXe-88 sent to me, so I decided to get some VSSOP-10 breakouts and order the INA228, INA237 and INA238 to work on.  While the empty prototype boards are available everywhere, it seems that the INA2xx chips are out-of-stock with a delivery in MAY 2022!! I checked digikey and mouser with no luck.  The only source I found for a INA228 was the evaluation kit from TI for €52 which I'm debating on getting so that I can get cracking on this, but that is a lot of money for something that I really don't need.
I'll keep this issue updated with my progress.

sebashb, 2021-12-20:
We have both bare INA228s and TI’s demo board. I think we can send either one or both  to you (on the case of the demo board we would need it back later on) if you want.

SV-Zanshin, 2021-12-20:
@sebashb - Thanks for the kind offer! I know I ordered INA228 bare chips and am going to see if I can find them after all; although I'm not very good at soldering such small devices I did make a reflow toaster-oven a while back and will see if that is still working to put the VSSOP-10 components onto a prototype board -- if I find the INA228. Otherwise I think I might just bite-the-bullet and order the evaluation board in order to save time.

jhardesty200, 2021-12-23:
@SV-Zanshin I also have a few extra INA228 on custom boards if you need some. =)

SV-Zanshin, 2021-12-23:
@jhardesty200 - Could I take advantage of that and get one INA228 on a board? I found one chip here, but don't have a VSSOP-10 to solder it onto and odds are good that with just one chip I'd mess up any soldering attempts... I'm currently in Germany - where would you be sending from?

jhardesty200, 2021-12-27:
@SV-Zanshin I am currently back home in California for the holidays and won't be back to my place until mid January I do have one INA228 and INA229 on hand as well as access to a rework station with a few VSSOP breakouts. I can get you the boards when I get back from California or I can send you the 228 that I have but I would like to do a little bit more testing before then or if I can get my hands on another bare chip to solder on to a breakout that I can test on. Sorry for the delay! =(

YuMERA, 2022-02-05:
Any progress in adding INA229 SPI ic to your library?

JAGX13, 2022-04-05:
Can anyone tell me if it's possible to use this library on Atmega 4809 (Uno wifi R2)? I want to use the INA228

Robocraze, 2022-04-05:
Hi Zanduino/INA,

Thank you so much for taking the time to connect with us!

Your ticket 1XJ0HK has been received. You can reply to the ticket via this email.

Please note, due to the large volume of support tickets, our agents may take up to 48 hours to respond to your query (it might take a bit longer on evenings and weekends). We thank you for your patience.

Thanks and Regards
Robocraze Support Center

This email is delivered by www.HelpDesk.com

Ticket ID: hd.1649195826907.1xj0hk.99a0b7ff