Zanduino/INA #78: Hi Zanduino,   Is it Possible to Integrate the INA237 ?  Our requirement is to support 85V .

ajay2110, 2021-03-10:
**Describe what the feature request is and if it solves an issue or adds functionality**
_A clear and concise description of what the problem is which would be solved by the feature request. Example "It would be helpful to add a function newFunc() to solve the problem of [...]"_

**Describe the solution you'd like**
_A clear and concise description of what you want to have changed or added._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

SV-Zanshin, 2021-03-10:
This is a similar request to #77 .  It looks like the INA237 won't be available on digikey.com until April.

SV-Zanshin, 2021-03-20:
Do you have an INA237 to test this on? I could write the software code, but can't test this at the moment.

SV-Zanshin, 2021-03-25:
The INA237 is still out-of-stock at Digi-Key, otherwise I would have ordered one to test

ftyers, 2022-02-15:
@ajay2110 if you still need this, please feel free to try out the PR #88.

UtkuDenizAltiok, 2025-01-02:
Did you solve the problem? I also need it for higher voltage range