Zanduino/INA #80: {"Add Short one-line Bug title"=>""}

michael-brade, 2021-06-03:
Hi, this is just to point out a copy&paste error. Here:

https://github.com/Zanduino/INA/wiki/alertOnBusOverVoltage()

In the second paragraph, it says: "...millivolts for the shunt voltage". It should be "bus voltage".

And the example has:
```
INA.alertOnBusOverVoltage(true;5000);
```
should be `(true, 5000);`, not a semicolon.

Thanks for a great library :-)

michael-brade, 2021-06-03:
Actually, while I'm at it: INA.cpp, line 902:
```
...when the bus current goes aboe...
```
should be
```
...when the bus voltage goes above...
```

SV-Zanshin, 2021-06-04:
Both errors corrected -thanks!

michael-brade, 2021-06-04:
Thanks, but I don't think so :stuck_out_tongue_closed_eyes: 

The first issue in the wiki still says "shunt voltage" instead of "bus voltage" in the second paragraph.

Actually, looking at the alert pages, they all say:

> The state will remain set until the voltage on the shunt goes below the threshold limit.

But the *Under-pages should say "goes above" and the shunt needs to be replaced with bus or power.

And all the alertOn*Voltage-wiki pages have the ; after the `true`.

Second issue had the typo that you fixed, but that wasn't the important bit -- it was `s/current/voltage/`, and it still says current instead of voltage in INA.cpp, line 902.

SV-Zanshin, 2021-09-29:
I missed this one until now. I've updated the files to reflect the correct program code in the Wiki and comments in the source code.

closed 2021-09-29 by SV-Zanshin