Zanduino/INA #81: Max expected amps < 1A

w407022008, 2021-06-20:
Hi, thanks for the nice library. I plan to use a power monitor to measure the power of a heating resistor, for example the ina226. Since the heating power is very low, I need to monitor a maximum circuit of less than 1 amp. I checked the ina226 manual and it doesn't seem to specifically state that the minimum current should be greater than 1 amp (maybe I didn't find it), so before I purchased it I came to ask if I could modify [this line](https://github.com/Zanduino/INA/blob/71b7dc4319d71b3eacd093f3182a8930b3435200/src/INA.cpp#L26) of code to make it usable by me.

SV-Zanshin, 2021-06-20:
The library is designed to use the best possible LSB, so just specify the 1A and it will use the minimum LSB value for the INA226

w407022008, 2021-06-20:
Hi, thx a lot for your reply.
What you said means that the minimum value of the maximum expected current is 1 Amp, and accordingly the minimum current resolution is 30.5 µA/bit (=1/2^15). May I ask if this is the way to understand it? Because I only see the formula for calculating the current_LSB based on the maximum expected current in the ina226 documentation, but I don't see the allowable range of the maximum expected current, so I would like to have confirmation of that. Actually 30.5 µA/bit of minimum current resolution is good enough for me. I'm just confirming if you know the answer with certainty. Thanks again.
Ze

w407022008, 2021-06-20:
Hi, After reading the documentation in detail, I might say that the maximum expected current is indeed an arbitrary customized parameter. Only the factors Shunt_Voltage_LSB and Bus_Voltage_LSB are fixed and respectively 2.5 µV/bit and 1.25mv/bit, while the factor Current_LSB is calculated based on the input maximum expected current. Current_LSB will be a function of Shunt_Voltage_LSB, i.e Current_LSB = Shunt_Voltage_LSB * 0.08192 / Maximum_Expected_Current / Shunt_Resistance. Therefore, if we use a 4 ohm shunt resistor, it is best to set the maximum expected current as 20 mA, so that the two binary values are close to each other. Please point out if there is any misunderstanding.

SV-Zanshin, 2021-09-29:
Yes, your understanding is spot on!

closed 2021-09-29 by SV-Zanshin