Zanduino/INA #87: Power does not show negative values.

Oleg-Sob, 2022-01-16:
Why INA226 with negative current readings, the power shows positive values. Should be negative too.

[image]

Please tell me how can I fix this.

SV-Zanshin, 2022-01-16:
How are you calcultaing "Power"? Are you using "getBusMicroWatts()"?

Oleg-Sob, 2022-01-16:
I output to Home Assistant with a script via EspHome.

i2c:
  sda: D2
  scl: D1

sensor:
  - platform: ina226
    address: 0x40
    shunt_resistance: 0.0005 ohm
    current:
      name: "INA226_1 Current"
      accuracy_decimals: 1
    power:
      name: "INA226_1 Power"
      accuracy_decimals: 0
      filters:
       - lambda:  return x*2.047;
    bus_voltage:
      name: "INA226_1 Bus Voltage"
      filters:
       - lambda: return x*2.047;
    shunt_voltage:
      name: "INA226_1 Shunt Voltage"
      accuracy_decimals: 0
      filters:
       - lambda: return x*107800;
    max_current: 100A
    update_interval: 10s

SV-Zanshin, 2022-01-16:
I have never used Home Assist of EspHome so I'm not sure what that script does. How does it calculate power?

SV-Zanshin, 2022-01-16:
I've updated the "getBusMicroWatts()" function to return negative Watts, but I don't know if that is the function you are using.